muart-group/esphome #11: [PROPOSAL] Kumo Emulation Mode

KazWolfe, 2024-05-12:
It seems likely that users would use this project as a replacement to a Kumo Cloud unit, seeing as any integration created by the community would likely be (a) cheaper, (b) better, and (c) more supported. As such, it would be nice if the library could provide other Kumo-related comfort features, including:

- Dual setpoints in auto mode
- Blocking the MHK2 from creating schedules
- Humidity/Battery Reporting

To that end, the project will need some ability to emulate a Kumo Cloud device.

### Detecting a Kumo Cloud

Kumo Emulation Mode should not operate fully when an _actual_ Kumo Cloud is connected. As such, it is imperative that Kumo Emulation Mode only be active when an MHK2 is connected _and_ no upstream Kumo is found.

To detect the presence of an MHK2, we only need to check for an [Identify packet](https://github.com/Sammy1Am/mitsubishi-uart/wiki/(0x41)-Set-Request#type-0xa7---mhk2-identify) coming from the configured thermostat UART. 

To detect the presence of a Kumo Cloud, we need to forward the identify packet to the upstream device. If the device responds with a 0x61 response, we know that there's an upstream Kumo device and that we should not attempt to replicate all features, or make certain behavioral choices (e.g. resolving auto mode requests). In this mode we will likely need to forward most packets as-is, but some changes may be necessary to handle things like passing the aforementioned dual set-point on for resolution. 

If a Kumo Cloud is *not* detected, then we have free reign to emulate one fully and handle any such responsibilities accordingly.

paravoid, 2024-07-22:
@KazWolfe wasn't the PR for this merged a while ago? Is there anything left to do here?

KazWolfe, 2024-07-23:
This isn't fully enabled yet since there's still quite a bit we don't know about the Kumo protocol, but I am happy enough to close it.

closed 2024-07-23 by KazWolfe