muart-group/esphome #51: Failed external_components

Olgmo, 2024-06-20:
After last update the something has changes to source. Please update wiki.

`external_components: [source /config/kaiteki.yaml:63]
  - source: github://muart-group/esphome@dev
    components: 
     
      Could not find __init__.py file for component mitsubishi_uart. Please check the component is defined by this source (search path: /config/.esphome/external_components/4ca37e81/esphome/components/mitsubishi_uart/__init__.py).
      - mitsubishi_uart`

Sammy1Am, 2024-06-20:
Sorry, will update as soon as able.  Some documentation available here in the meantime:
https://deploy-preview-3860--esphome.netlify.app/components/climate/mitsubishi_itp

Component is now called `mitsubishi_itp`.

Olgmo, 2024-06-20:
NOTE!
It should be **ITP** instead of IPT:
correct syntax

climate:
  - platform: mitsubishi_itp

Sammy1Am, 2024-06-20:
Sure enough, thanks, I'll fix that too when I get to the wiki.

Sammy1Am, 2024-06-21:
Sorry again about the offset update to documentation and component.  Generally try to avoid big breaking changes like that, but that one just had to be done.

Docs are updated, thanks for the reports.

closed 2024-06-21 by Sammy1Am