muart-group/esphome #52: UART config validation error

awkaplan, 2024-06-21:
Looks like 1f89bce5d019e98dd1e62297fbc3a31ced735dd8 includes a kwarg that the i2c __init__.py does not expect:

Trace:

```
INFO ESPHome 2024.6.1
INFO Reading configuration /config/office-hvac.yaml...
ERROR Unexpected exception while reading configuration:
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/esphome/esphome/__main__.py", line 1079, in main
    return run_esphome(sys.argv)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/__main__.py", line 1057, in run_esphome
    config = read_config(dict(args.substitution) if args.substitution else {})
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 1093, in read_config
    res = load_config(command_line_substitutions)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 947, in load_config
    return _load_config(command_line_substitutions)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 937, in _load_config
    return validate_config(config, command_line_substitutions)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/esphome/esphome/config.py", line 861, in validate_config
    result.run_validation_steps()
  File "/esphome/esphome/config.py", line 144, in run_validation_steps
    task.step.run(self)
  File "/esphome/esphome/config.py", line 740, in run
    self.comp.final_validate_schema(conf)
  File "/config/.esphome/external_components/4ca37e81/esphome/components/mitsubishi_itp/climate.py", line 296, in final_validate
    schema = uart.final_validate_device_schema(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: final_validate_device_schema() got an unexpected keyword argument 'uart_bus'
```

Config:

```
external_components:
  - source: github://muart-group/esphome@dev
    components: [ mitsubishi_itp ]

uart:
  - id: hp_uart
    baud_rate: 2400  # Note: May be 9600 on some systems!
    parity: EVEN
    rx_pin:
      number: GPIO17
    tx_pin:
      number: GPIO16

climate:
  - platform: mitsubishi_itp
    uart_heatpump: hp_uart
```

Sammy1Am, 2024-06-21:
Ah, hmm...  I wonder if this is a change in ESPHome 2024.7...  In the meantime you can target `@e60f6d4` instead of `@dev` to skip the validation commit.

paravoid, 2024-06-21:
This needs https://github.com/esphome/esphome/pull/6923, which is merged in ESPHome dev. If you don't want to run dev, or wait for 2024.7, you can use `external_components`  to pull in `uart` specifically.

awkaplan, 2024-06-21:
I can confirm, setting the following works around this issue:

```
external_components:
  - source:
      type: git
      url: https://github.com/muart-group/esphome
      ref: dev
    components: [mitsubishi_itp]
  - source:
      type: git
      url: https://github.com/esphome/esphome
      ref: dev
      path: esphome/components
    components: [uart]
```

paravoid, 2024-07-17:
2024.7.0 has been released, so I think we can close this issue.

closed 2024-07-18 by Sammy1Am