muart-group/esphome #65: Fahrenheit temperature representation issue.

galmeida, 2024-12-14:
Hi, my HA is set to use Fahrenheit, when I set the target temperature to certain values, the value will change after a few seconds by 0.5 degree, looking at the code, this seems to be a limitation of the protocol ([mitp_utils.h:L23](https://github.com/muart-group/esphome/blob/dev/esphome/components/mitsubishi_itp/mitp_utils.h#L23)) and not any issue with the code itself. Despite that, I don't recall seeing the same behavior on the MHK2 thermostat, I wonder if there is any way to workaround this?

Log sample demonstrating the issue:
```
[21:01:14][D][climate:396]: 'Climate' - Sending state:
[21:01:14][D][climate:399]:   Mode: HEAT
[21:01:14][D][climate:401]:   Action: HEATING
[21:01:14][D][climate:404]:   Fan Mode: AUTO
[21:01:14][D][climate:419]:   Current Temperature: 22.00°C
[21:01:14][D][climate:425]:   Target Temperature: 21.67°C
[21:01:18][I][mitsubishi_itp:277]: Received temperature from Thermostat of 22.000000. (Current source: Thermostat)
[21:01:21][D][climate:396]: 'Climate' - Sending state:
[21:01:21][D][climate:399]:   Mode: HEAT
[21:01:21][D][climate:401]:   Action: IDLE
[21:01:21][D][climate:404]:   Fan Mode: AUTO
[21:01:21][D][climate:419]:   Current Temperature: 22.00°C
[21:01:21][D][climate:425]:   Target Temperature: 21.50°C
```
71F → 21.67C → 21.5 → 70.7F → 70.5

Sammy1Am, 2024-12-14:
These issues actually drove me to change all my IoT devices to °C, even though I think it's objectively a worse scale for climate control use. 😵‍💫

The MHK2 only shows whole-degrees F (IIRC), so in your example it would round back up from 70.7F to 71F.  Because the heat pump internally uses half-degrees C intervals, 70.5F _and_ 71F in HA both set the unit to 21.5C.  So it's not that the target temperature is really changing, both of those targets in F are equivalent and 70.5 is technically closer to the real setpoint than 71.

Over/under runs get even weirder, because with rounding, going over the setpoint by 0.5°C might actually display as 2°F over your setpoint.

One potential workaround is to force the HA climate control to show only whole-degrees F instead of increments of 0.5.

Otherwise I'm not sure much can be done.  Since internally the equipment is set to 21.5°C, there's not an easy way to tell if you want that displayed visually in HA as 70.5 (the closest half-degree F value) or 71 (less correct, but a whole-degree).

galmeida, 2024-12-15:
> One potential workaround is to force the HA climate control to show only whole-degrees F instead of increments of 0.5

That's actually a simple and effective idea, but like you, I just changed everything to Celsius =)

Thanks.