muart-group/esphome #66: Outdor temperature reported as -63.5 when compressor is off

rene-dev, 2025-01-18:
Outdor temperature reported as -63.5 when compressor is off, it is only reported correctly when the compressor is running.
Maybe better to report as unavailable?
this is on a MSZ-HR unit.
[image]

Sammy1Am, 2025-01-24:
Outdoor temperature has definitely been an inconsistent sensor for everyone.  I don't know off-hand if anyone else has seen this exact behavior though (i.e. a specific value when the compressor is off).

You should be able to achieve the desired behavior with **something like** this (I haven't actually tested this, but to get you in the right direction:
```yaml
sensor:
  - platform: mitsubishi_itp
    outdoor_temperature:
      # Outdoor temperature as reported by equipment
      name: "Outdoor Temperature"
      availability: !lambda |-
        return id(outdoor_temperature).state != -63.5
```

svmaris, 2025-01-24:
I'm having the same "issue" with my units (also MSZ-HR, in The Netherlands) and it has been previously discussed in [another thread](https://github.com/muart-group/esphome/issues/45#issuecomment-2188449087) as well. When the compressor is off, the value of the sensor is reported as 0x01, which translates to -63.5°C (`temp = (x-128)/2`).

My guess is that this sensor is used by the unit to detect when to run a defrost cycle. It's not a very accurate way to measure the outdoor temperature in my case. I've also seen measurements that were way higher than the actual outdoor temperature (maybe during a defrost cycle? Can't remember).

Sammy1Am, 2025-01-24:
I hadn't checked for a while, but just confirmed that mine reports temperature even when the compressor is off.  To be sure, I turned the system (all three heads) to off as well, but it's still reporting the temp.  I have an SVZ-KP air handler and a MXZ-60NA2 outdoor unit, however I do also have a PAC-MKA52BC branch box.  I've previously suspected that the branch box is the reason I don't get a compressor frequency reading, but maybe it somehow also keeps the outdoor temp sensor active?

(The spikes do line up with defrost cycles)

[image]

It seems like behavior is slightly system-dependent, but I don't see any reason it wouldn't be safe to assume that a `0x01` value is "unavailable" rather than -63.5°C (Until someone from McMurdo Station opens a ticket, at least 😉).  Will keep this open to remember to add that next time I'm in the code.  The snippet of ESPHome config above should be close to usable as a stop-gap until then.

ignisf, 2025-04-07:
Confirming outdoor temperature on MSZ-AP50VGK is incorrect whenever the unit or the compressor are off.

Sammy1Am, 2025-04-07:
I think this is consistent enough to warrant a fix in the code (unlikely anyone will be actually experiencing -63.5.  Will track fix using this issue (no ETA at the moment)

ignisf, 2025-04-07:
I mean... I guess you could try to use these heat pumps at -63.5C. :)

Sammy1Am, 2025-08-02:
Optimistically closing since it's in `dev` in the other repo. Reopen if issue persists in that codebase.

closed 2025-08-02 by Sammy1Am